diondokter/device-driver #6: Silently writes 0 if not using chaining.

Dirbaio, 2021-04-19:
If you do this, the correct value is written.

```rust
regs.mode_ctrl().write(|w| w.mode_ctrl(Mode::Lowpower).sleep_durn(SleepDuration::_1S))
```

However, when doing this, it writes `0`.

```rust
regs.mode_ctrl().write(|w| {
    w.mode_ctrl(Mode::Lowpower);
    w.sleep_durn(SleepDuration::_1S);
    w
});
```

This was quite surprising to me, since the latter works correctly with svd2rust, and the API generated by this crate is very similar. I always write in the 2nd style, I find it more readable to have one field per line.

diondokter, 2021-04-19:
Really? I'd consider that a bug. Good of you to post :)

I'll look into it

diondokter, 2021-04-19:
Right, so we can see what happens when the code is expanded:
```rust
/// Mask bit for output 1
pub fn mask_1(mut self, value: Bit) -> Self {
    use device_driver::bitvec::prelude::*;
    use device_driver::bitvec::view::AsBitsMut;
    let raw_value: u8 = value.into();
    self.0.as_bits_mut::<Lsb0>()[5..=5].store_be(raw_value);
    self
}
```

It's not working with references, but with instances. (Been a while since I looked at this)
I see no reason why this couldn't be done with &mut references.

PR #7 (2021-04-19): All w interactions are now through mut references
Fixes #6
commit 04d452d7 (2021-04-19): Fixes #6

Added some test cases in the example to check this usage in the future
closed 2021-04-19 by diondokter