diondokter/device-driver #8: Support debug prints on `R`

korken89, 2021-04-29:
Hi,

For example when I read a register it would be really nice in development to be able to add debug prints of the register.
Eg, lets say I have a register with 2 flags:

```rust
implement_registers!(
    MyChip.registers<u8> = {
        status(RO, 0, 1) = {
            flag1: u8 as Bit = RO 0..=0,
            flag2: u8 as Bit = RO 1..=1,
        },
        // ...
    }
}
```

Debug-printing with `{:?}` I'd expect something like `status::R { flag1: Set, flag2: Cleared }`, and with  `{:#?}`

```
status::R { 
    flag1: Set, 
    flag2: Cleared 
}
```

It would also be preferable if one use custom enums that there are also debug printed with correct variants.

diondokter, 2021-04-29:
Right, this is actually more elaborate than I thought because I'd just print the buffer in hex format.
But this is clearly better! So we can do both :)

How should we deal with field types that don't implement Debug themselves? Ideally we'd detect that and print the raw numbers, but I don't think we can...
Or maybe we should require all fields to implement Debug? But that'd be kinda strict and not always necessary...

korken89, 2021-04-29:
Maybe have a flag to request debug that puts a requirement on `Debug` and else just print it in HEX?
Then the user can choose to do it or not :)

diondokter, 2021-04-29:
So I imagine we'll like more of these extensions and restrictions the user can opt in to.
Could try to emulate (only syntactically) what `derive` does and maybe call it `generate`.

```rust
implement_registers!(
    MyChip.registers<u8> = {
        #[generate(Debug)]         // <- New!
        status(RO, 0, 1) = {
            flag1: u8 as Bit = RO 0..=0,
            flag2: u8 as Bit = RO 1..=1,
        },
        // ...
    }
}
```

I try to avoid completely alien syntax in the macro and I think this would feel familiar and also distinct enough.

korken89, 2021-04-29:
Looks good to me!

diondokter, 2021-04-29:
@korken89 I added support for it!
Would you mind trying it out before I merge it?

PR #10 (2021-05-01): Added #[generate(Debug)] feature
Closes #8 

You can now do this:

```rust
implement_registers!(
    MyDevice.registers<u8> = {
        // (comments must go first)
        #[generate(Debug)]              // <- new!
        id(RO, 0, 3) = MSB {
            manufacturer: u16:LE as Manufacturer = RW 0..16,
            version: u8:NE = RO 16..20,
            edition: u8:BE = RO 20..24,
        },
    }
);
```

Then we can:

```rust
let id = device.registers().id().read()?;
println!("{:?}", id);
```
```
id::R { raw: 0x010065, manufacturer: Ok(CarmineCrystal), version: 6, edition: 5 }
```

The gener [... cut]
closed 2021-05-01 by diondokter