diondokter/device-driver #12: Feature request: Strict mode

korken89, 2021-05-01:
A way to say to the macro that I do not want a fallible interface for register decoding (enums).
So if it can detect that an fallible interface would be generated it should scream at me, or maybe a trait error. :)

The infallible interface would also be nice if it could be related to the bit size, e.g. a 2 bit field that is linked to a 4 variant enum (or less + FromPrimitive implemented).
Then it is known that the entire value span is well defined.

korken89, 2021-05-09:
I was playing around a little, and the following crate gave a lot of good underlying support for enums (and a lot more): https://docs.rs/modular-bitfield

diondokter, 2024-06-01:
This is now sorta kinda implemented by #24, so I'm gonna close this issue :)
(There's no detection because that's kinda hard the way everything is setup, but you can say it's strict and get a type error when it isn't)

closed 2024-06-01 by diondokter