diondokter/device-driver #13: Aliasing RO/WO bits cause codegen failure with `generate(Debug)`

korken89, 2021-05-08:
I have noticed that generating dual definitions within a register works fine, for example:
```rust
        // #[generate(Debug)]
        tuning_r(RW, 0x1d, 1) = {
            dl_tune: u8 = RW 4..=7,
            lead_lag: u8 = RO 3..=3,
            cagainov: u8 = WO 3..=3,
            integlen: u8 = RW 2..=2,
            integgain: u8 = RW 0..=1,
        },
```

The bit which has different meaning when writing and reading is generated correctly in the `R` and `W`.
However when I add the `#[generate(Debug)]` the debug tuple is not correctly constructed with the debug info for the WO bit is in the `R` debug print impl:

```rust
            impl core::fmt::Debug for R {
                fn fmt(&self, f: &mut core::fmt::Formatter<'_>) -> Result<(), core::fmt::Error> {
                    f.debug_struct("tuning_r::R")
                        .field(
                            "raw",
                            &device_driver::utils::SliceHexFormatter::new(&self.0),
                        )
                        .field("dl_tune", &self.dl_tune())
                        .field("lead_lag", &self.lead_lag())
                        .field("cagainov", &self.cagainov())
                        .field("integlen", &self.integlen())
                        .field("integgain", &self.integgain())
                        .finish()
                }
            }
```

So to me it seems that the `#[generate(Debug)]` does not handle WO bits correctly as they should not be in the debug print.
What do you think @diondokter ?

diondokter, 2021-05-08:
Ah right! It's pretty obvious spelled out like this. I did not think about checking the field RW state so it currently tries to show all fields regardless.

Gonna need an extra macro to fix 😅
But this is also doable. Will look into it.

Thanks for spotting it!

diondokter, 2021-05-17:
So, I had some time (and motivation) today. Sadly the way I thought I was gonna solve this doesn't work.
I may have to do a really ugly workaround. But that's ok if I'm gonna port this to a proc macro after this...

Sorry for taking so long :(

korken89, 2021-05-23:
No problem!

diondokter, 2024-01-07:
Crate has been overhauled. Issue no longer relevant

closed 2024-01-07 by diondokter