diondokter/device-driver #18: Broken links in crates.io README file

chrysn, 2024-01-07:
The links on https://crates.io/crates/device-driver/0.5.0 for the various examples go 404 -- they contain twice the `/device-driver/` path component.

I'd file this as a PR, but I have no clue as to which magic makes the relative references in there work on crates.io in the first place – is there some regeneration of the README at release time involved? (Might be related to the tags not being set).

If there is no file mangling magic, please let me know and close this issue; then, this must be some crates.io mixup, maybe as part of processing the readme.workspace attribute.

diondokter, 2024-01-07:
Argh, yeah I see...

It does do some mangling. Idk how to solve that except for duplicating the readme...

diondokter, 2024-01-07:
They do work on GitHub, but in the crate the root changes to the crate folder.

diondokter, 2024-01-08:
Thanks! I fixed it in the last release with hard links instead of relative links

closed 2024-01-08 by diondokter