diondokter/device-driver #35: More control over case conversions?

alexmoon, 2024-08-01:
Right now converting names from pascal case to snake case can lead to some undesirable word boundaries. For example `I2cSettings` turns into `i_2_c_settings` in snake case and `I2CSettings` (note the capital `C`) in pascal case. I'm not sure what the best solution would be to give the user more control over those conversions. One option would be to use an unambiguous casing as the input in all cases (such as snake case). Another option would be to let the user optionally specify the casings somehow.

I would consider this a pretty minor issue, but something worth thinking about; particularly if you move forward with something like #34.

diondokter, 2024-08-01:
Yeah that's true. Can be annoying at times. I made it like this so it's easy to copy from datasheets which are not always consistent.

I want to have some global settings and the casing can be part of that. The current could be called 'best guess', but others could maybe specify that the input case is always Pascal.

diondokter, 2024-08-23:
So what I've settled on now is being able to specify the word boundaries yourself fully in the global config: https://github.com/diondokter/device-driver/blob/4c2a0414f68a99f9e7d10759556484d5aaac54d8/generation/src/dsl_hir/mod.rs#L20-L34

You can specify the boundaries as an array: https://docs.rs/convert_case/latest/convert_case/enum.Boundary.html
Or as a string: https://docs.rs/convert_case/latest/convert_case/enum.Boundary.html#method.list_from

PR #40 (2024-10-22): Compilify
Complete rebuild.

Closes #42 
Closes #35
Closes #34  
Closes #33 
Closes #31 
Closes #30 
Closes #29
Closes #21

Posting this since we're nearing finishing.

Todo:

- [x] Restructure as a compiler
- [x] Better name normalization
- [x] Address overlap checks
- [x] Field overlap checks
- [x] Commands can have fields
- [x] Per field access setting
- [x] Better enum generation
- [x] Nicer type conversion
- [x] Redo field writer struct generation
- [x] Add bit manip traits to field writer struct
- [x] Add blocks for everything
- [x] Add refs for everything
- [x] Add repeat for everything [... cut]
closed 2024-10-22 by diondokter