diondokter/device-driver #38: Add option for multi register read/write

diondokter, 2024-09-14:


diondokter, 2024-11-29:
Ok I'm gonna pause this effort.
I've now spent many hours on this.

You'd think having to deal with all the different register types would be the problem. But no. That's fine.
The real issue that we have an unknown number of registers which all have an unknown bitsize.
To deal with this we need const generic expressions which are still very unstable.

enbyted, 2025-01-06:
I don't know if this is a correct issue for raising this concern, but it seems related.
Currently the `read_all_registers` function is generated as just calling read on each register in sequence, however that may not always be valid.
I'm not sure how this can be addressed properly, probably via this ticket's proposal of arbitrary multi-register read-write (which seems to be blocked on const generics).
However, it would be cool to at least have the ability of suppressing the generation of that method, or adding documentation to it.

### Technical details of an example where that is relevant

For example, for VS1003 audio codec each register access, even read, will cause a data request (dreq) line to go low and you are not supposed to execute any command during that period. 
It complicates matters that that dreq may not fall immediately after receiving the command.

Now, the datasheet claims that this delay is not relevant for external microcontroller and that the busy-time for reads is too short to notice and thus can be ignored (which is technically true, more on that below), however obviously that was not written with 100+ MHz chips in mind.

In my abstraction I chose to monitor the dreq pin before doing an SPI transfer and returning an error code in case it's low. This works our fine if the user of the PAC library is careful (as they should - there are always a lot of invariants to hold to properly use a peripheral after all, so people tend to read the docs). With that in mind the generated method is clearly wrong.

I do not want to add a delay after a register operation as in typical operation that would be unnecessary (and would require the interface to take a delay source or lock the user to some specific time-provider crate).

#### Offtopic

My chosen behavior is not actually 100% correct for that device, becuase what the dreq being low means is that you are not allowed to **finish** a transaction, not start it. Thus, a most performant implementation would send 15 bits (as register access takes 16 bit header) and hold off on sending the 16th clock and bit until dreq rises. Obviously, that is incompatible with standard SPI hardware and typical embedded-hal abstractions, hence my chosen method.

I may actually drop the dreq handling from the pac crate completely and just move the management of this line to the driver crate instead. I haven't made up my mind yet. That would make the `read_all_registers` work in most cases. Most cases, because when the VS1003 is running a plugin, the timing requirements may change, even for register reads.

diondokter, 2025-01-06:
@enbyted hi, interesting problem. I'm open to ideas on fixing it.

That said, the function is only really meant for debugging purposes. So if it doesn't quite work for everyone, that's fine in my books.

My question is, why return an error when dreq is low? If it's supposed to be really short, maybe just do a short blocking wait on it? That would fix it as well. That way you're only waiting on it when you really have to.

diondokter, 2025-01-07:
Ok, well... Not much I can do then. I see your problem.

The only other suggestion I can make is to make the interface blocking, but to expose an `is_busy` method.
Then the driver will always work, but people who need to optimize still have the option to do so.
This way you're also not using the errors for normal happy path control flow, which is nice too.

Code doing things in parallel would then become:
```rust
loop {
  loop {
    if !device.is_busy() {
      let value = unwrap!(device.reg().read());
      do_something_with_value(value);
    }

    do_something_else();
  }
}
```

Code that is fine with blocking can just read the register without having to do a lot of boilerplate.
This would also make the `read_all_registers` function work.

But that's up to you. Unless you have a proposal to add a readiness concept to the crate, I don't think I can help much more.

enbyted, 2025-01-07:
> Unless you have a proposal to add a readiness concept to the crate, I don't think I can help much more.

Actually, I've been thinking about that, as different devices expose their ready state very differently (sometimes even in many ways and you have to decide which one you want - ex. ready pin, vs. pooling a status register).

It might be nice to have a standardized `is_busy()`, `is_busy_async()`, `wait_until_ready()` and `wait_until_ready_async()` methods on the generated interface, which is then translated into whatever code is supplied by the interface (via a new trait?).

For now, that is exactly what I'm doing in addition to providing the error return.

> The only other suggestion I can make is that you do make the interface blocking, but also expose an is_busy method.

It wouldn't work in case of this particular device. In the MCU's clock speed is at least half as fast as the device's (50 MHz typically) the code might notice the fact that it takes some time for the dreq pin to fall, leading to a particularly sinister situation where:

```rust
device.reg().read().unwrap();
while device.is_busy() {}
device.reg().read().unwrap();
```

will panic on the third line with `ErrorEnum::Busy`... learnt that the hard way.

And I can't just wait for the dreq pin to fall after SPI transaction as it is not defined how fast the function will return (especially in a RTOS environment) and thus I might miss the low period completely.

Anyhow, that is not your concern, sorry for the offtopic. 
Hopefully at least it provides some insight for you when you will design any future APIs, so you can take such weirdness into account, or not, at your own discrection :smiley:

diondokter, 2025-06-20:
Interesting feature I hadn't considered yet:
Some people will want one transaction where both writes and reads happen.

`Multi([write(foo), read(bar)])`
This is even crazier to support...

diondokter, 2025-08-31:
Added to milestone 2.0 because we can do breaking changes and this deserves another shot before that release.

I previously got stuck with combining multiple fieldsets into one. But maybe the RegisterInterface api can be changed so it can receive multiple 'operations' just like the embedded-hal SpiDevice can.

alexmoon, 2025-10-07:
> Also, not everything can be done in one transaction due to hardware constraints, so we'll need limits for that too:
> 
>     * Registers may need to be adjacent in address (address space or logically)
> 
>     * Mixing reads and writes may or may not be supported

I think that the RegisterInterface::transaction impl could just return some sort of Error::NotSupported error if an invalid transaction is requested for a particular device. It would be nice to capture those constraints at compile time but I'm not sure it's necessary.

diondokter, 2025-10-15:
> How exactly does `execute` come up with whatever buffer space it needs? I guess it should be possible with whatever const generics are currently available, but all ways in which I see it in my mind are quite messy.

Yeah, I've tried putting everything in one big array using const generics, but it's not ready for that yet. That's what the second post in this thread is about.

> What about an iterator like interface for fields of the same input/output type?
> 
> ```rust
> fn execute<T>(&mut self, plan: impl IntoIterator<Item = PlannedOp<Result = T>>): impl Iterator<T> {}
> ```

I see your vision. For most C HALs this would totally suffice. However in rust's embedded hal, you got things like [SpiDevice](https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/trait.SpiDevice.html) that requires all operations up front. So even if you do it through an iterator, the interface implementation will need to collect it again somehow.

This is still an issue with the renewed interface trait idea I wrote here: https://github.com/diondokter/device-driver/issues/38#issuecomment-3369262922

alexmoon, 2025-10-16:
> This can be simplified btw. Here the transactionplan is basically a wrapper around a tuple. So why not:
> 
> let plan = (device.foo().plan_write(), device.bar().plan_read(), device.quux().plan_read());
> let (bar, quux) = plan.execute(device, |foo| foo.set_val(true)).unwrap();

Yes, in my version I impl the Transaction trait for Option<T> and tuples up to a certain number of elements.

diondokter, 2025-10-17:
> Reason being, that having everything up front is not always desired.

Do you know of a device that demonstrates this?

> I wouldn't necessarily discount interfaces that require features not supported by common traits.

If there's any main goal in this project, it's that you should be able to write platform independent drivers in Rust. That means the embedded-hal traits are critical. So IMO, anything that gets in the way of that is automatically discounted.

> Albeit it depends on nightly, and worse yet, it depends on incomplete [generic_const_exprs](https://github.com/rust-lang/rust/issues/76560) feature.

That's what I mean with that Rust is not ready yet. Having a nightly feature is one thing, but one that depends on `incomplete_features` is a whole other thing.

> I guess rust's embedded world is kind of used to being on nightly for the most part.

This used to be the case, but for the last ~1.5-ish years I've been using stable.

enbyted, 2025-10-17:
> > That is a use case requirement, not a device requirement though. Similarly, when you're developing on a device with 4-8 kB of RAM and you just can't spare the 512B to buffer 8 consecutive operations (and you have bus driver that can hold a transaction open).
> 
> This is something you could already build, just out of band of the device-driver toolkit. You're able to add functions, so you could add a `start_transaction()` and `end_transaction()`.

True, but it can become tricky with non-trivial / custom devices that take more than just the bus to drive. That would be out of scope of this crate probably, though.

> > How about passing in a buffer to work with and the ability for the driver to fail at runtime due to too small buffer?
> 
> That's similar to what I wrote here: [#38 (comment)](https://github.com/diondokter/device-driver/issues/38#issuecomment-3328767386) 

Ah, I forgot that was already proposed, sorry.

> it just doesn't support mixed reads and writes.

Why not? Writes encode data into the buffer before the transaction, reads decode data from the buffer afterwards. For most reads you anyway are doing a mixed write-read operation on the bus, to present the register address. If your bus driver can exchange data in the buffer (i.e. write byte and the replace it with the byte just read) - cool, otherwise the driver has to split the provided buffer down the middle. 

I don’t see why it wouldn’t work. It wouldn’t support RMW, but none of the proposals so far would. It’s a fundamental limitation of the approach we’re considering here and if you need that and batching you probably need more customization than a general-purpose toolkit can provide anyway. 

> I still like that approach

I like this one too. It seems most generic to me, as it can support both static (via tuples) and dynamic (via iterators) approaches while using mostly the same machinery. 

I might have a go at a semi-serious mockup tomorrow, if time permits.

diondokter, 2026-01-04:
Alright, update time: I came up with an idea that doesn't require users to allocate a buffer up-front.
#184 is my draft.

Here's what it looks like to use: https://github.com/diondokter/device-driver/blob/9fa1f6aa305f7ebefa01bbb0a0c55b10722cb7b3/device-driver/tests/basic-register.rs#L82-L126

I'm not done yet. I want to change some things and add some more features.
Feedback is helpful!

FYI, this does not implement mixing reads and writes in one transaction. It's a different problem (even though it overlaps!).
But this solution doesn't preclude anything like that in the future.