diondokter/device-driver #45: ReadOnly (RO) registers incorrectly generate setter functions

badrbouslikhin, 2024-11-11:
**Context:**
According to the documentation, field access modifiers should generate getters and setters based on their read/write capabilities. Specifically:
- Read access should generate a getter with the field name
- Write access should generate a setter with the `set_` prefix

However, I've observed that fields marked as `RO`/`ReadOnly` are incorrectly generating setter functions, despite being read-only fields that should only have getters.

**Expected behavior:**
- `RO`/`ReadOnly` fields should only generate getter functions
- No setter functions should be generated for read-only fields

**Actual behavior:**
- `RO`/`ReadOnly` fields generate both getter and setter functions
- Setter functions are being generated when they shouldn't be

**Example**
On the [S2LP driver](https://github.com/diondokter/s2lp/tree/main), [`MC_STATE1`](https://github.com/diondokter/s2lp/blob/36684eaa45be188d58f1668558adb866a06dd25a/device.yaml#L1775C1-L1775C10) is declared as a `RO` register. But setters get generated:
[image]

diondokter, 2024-11-13:
Hi, thanks for the report!

The system is working as intended though. (Maybe the intend is wrong, that's possible!)

There are two access specifiers: For the register and for the fields.
The one you see in S2-LP is a register access specifier. And since the default access for fields is RW, the setters are still generated.

Now maybe the fields should be influenced by the access of their register?
I guess that'd be logical. However, that could break 'ref' objects who share the same fields but might have a different register access specified.

So not sure how to improve that.

badrbouslikhin, 2024-11-13:
Understood. I can't think of an easy fix for this.
Maybe just document it for now?

Do you want me to open a PR to add this to the fields section of the book:
> Registers and fields have independent access specifiers. 
> By default, fields are set to RW (Read-Write) access, so setter functions are generated automatically. 
> Register access settings do not affect field access, but changing this could make the system more consistent. However, doing so might break ref objects, which often share fields but may have different access requirements at the register level.

closed 2024-11-13 by diondokter

badrbouslikhin, 2024-11-13:
Perfect, thanks!