diondokter/device-driver #50: Force CamelCase for `--device-name`

felipebalbi, 2024-12-12:
Could we have `device-driver-cli` force upper camel case for the driver name? Should be doable with the diff below:

```diff
diff --git a/cli/Cargo.toml b/cli/Cargo.toml
index 0ca4726..cb5d31c 100644
--- a/cli/Cargo.toml
+++ b/cli/Cargo.toml
@@ -13,6 +13,7 @@ readme = "README.md"

 [dependencies]
 clap = { version = "4.5.20", features = ["derive"] }
+convert_case = "0.6.0"
 device-driver-generation = { version = "1.0.0-rc.1", path = "../generation" }
 prettyplease = { version = "0.2.22", features = ["verbatim"] }
 syn = { version = "2.0" }
diff --git a/cli/src/main.rs b/cli/src/main.rs
index 5e875cf..34ef36a 100644
--- a/cli/src/main.rs
+++ b/cli/src/main.rs
@@ -1,4 +1,5 @@
 use clap::Parser;
+use convert_case::{Case, Casing};
 use std::{ops::Deref, path::PathBuf};

 #[derive(Parser, Debug)]
@@ -35,19 +36,19 @@ fn main() {
     let output = match extension.deref() {
         "json" => device_driver_generation::transform_json(
             &manifest_contents,
-            &args.device_name.to_string(),
+            &args.device_name.to_case(Case::UpperCamel),
         ),
         "yaml" => device_driver_generation::transform_yaml(
             &manifest_contents,
-            &args.device_name.to_string(),
+            &args.device_name.to_case(Case::UpperCamel),
         ),
         "toml" => device_driver_generation::transform_toml(
             &manifest_contents,
-            &args.device_name.to_string(),
+            &args.device_name.to_case(Case::UpperCamel),
         ),
         "dsl" => device_driver_generation::transform_dsl(
             syn::parse_str(&manifest_contents).expect("Could not (syn) parse the DSL"),
-            &args.device_name.to_string(),
+            &args.device_name.to_case(Case::UpperCamel),
         ),
         unknown => panic!("Unknown manifest file extension: '{unknown}'. Only 'dsl', 'json', 'yaml' and 'toml' are allowed."),
     };
```

diondokter, 2024-12-13:
Hi, yeah maybe this should just be enforced. Maybe not even be converted, but checked for PascalCase and then error if it's not. This is something the user just should do well.

No need for a PR, I'll add this in now.

diondokter, 2024-12-13:
Thanks for bringing it up!

PR #51 (2024-12-13): Enforce device name pascalcase & improve CLI errors
Closes #50