diondokter/device-driver #56: missing_docs lint is incompatible with generated code

enbyted, 2025-01-03:
Hi!

First of all, thank you for this crate - it certainly saves a lot boring boilerplate!
I tend to write my code with `#![warn(missing_docs)]` enabled by default. 

Unfortunately, the generated code for `pub mod field_sets` doesn't include any and there is no way to pass them. It's not clear to me what the docs should be and if there is any point in allowing the user to provide them. 

At any rate, since this module is also part of the public API of any crate using device_driver it probably should have docs.

enbyted, 2025-01-03:
That is the only one that warns.

There is also:
```rust
    pub(crate) fn interface(&mut self) -> &mut I {
        &mut self.interface
    }
```

Otherwise, everything else seems to have docs.

But I am only using the register features right now, got a device with relatively simple interface (VS1003 audio codec).

closed 2025-01-03 by diondokter

diondokter, 2025-01-03:
Thanks!

Fix released as 1.0.2 :)

PR #57 (2025-01-03): Add more docs to satisfy the missing docs lint
Fixes #56