diondokter/device-driver #58: Re-using definitions for families of chips

Wassasin, 2025-01-27:
I am currently writing a `device-driver` crate for the [Lumissil IS31FL3761 LED matrix driver IC](https://www.lumissil.com/applications/industrial/appliance/major-appliances/range-hood/is31fl3761). This IC is part of a family of chipsets that have a varying amount of current sinks \[[2](https://www.lumissil.com/applications/industrial/appliance/major-appliances/range-hood/is31fl3762), [3](https://www.lumissil.com/applications/industrial/appliance/major-appliances/range-hood/is31fl3763), [6](https://www.lumissil.com/applications/industrial/appliance/major-appliances/range-hood/is31fl3766)]. Furthermore a few registers are slightly different.

Hence I would like the crate I'm making to support all these devices, re-use the common definitions as-is, and change them slightly if necessary. Currently I'm using the YAML manifests to create this driver, but I am not married to the format. As it stands now, I would have to duplicate the entire YAML file for each chip variant, or create some kind of generator in build.rs that creates these YAML files.

Perhaps we can consider how we could provide a method in `device-driver` to handle these kinds of situations. Some ideas:
* include/merge YAML files.
* conditional 'compilation' within `device-driver` depending on macro invocation.

As an aside issue #47 would be useful for this chipset as well, as the current sink PWM values are big blob for which individual registers with fields are not useful, but I would still like to access them with an offset and size defined in `device-driver`.

diondokter, 2025-01-27:
There is limited support for `cfg` gates already. Maybe that's enough to be able to reach your goal? It doesn't apply to everything, but you can for example change which blocks/registers/fields/enum variants are active: https://github.com/diondokter/device-driver/blob/master/device-driver/tests/cfg-register.rs#L53

One hard part about this is that this crate does analysis as well. And right now it can't really know anything about cfg values since that's a property of the generated code and not of the generating code.

As for the memory object type, yeah I think that might be decent to add.

diondokter, 2025-05-16:
In anyone has suggestions for this, I'm all ears.

Couple of problems:
- How would you specify variations?
- What should be able to be varied? (Full registers, or also configs/repeat counts or fields or bit ranges of fields?)
- How do you select which version you are using?
  - Is this a choice when generating the driver?
  - Or should the generated code be able to work with cfg gates?
  - How about driver analysis like address overlaps?

diondokter, 2025-05-25:
Possible idea:

In KDL we can very naturally define a device:
```kdl
device Foo {
  // ...
}
```

In theory nothing is stopping us from adding more devices in one file:
```kdl
device Foo {
  // ...
}
device Foo2 {
  // ...
}
```

Here the first one could act as a template for the second one. `Foo2` could specify the modifications it makes on `Foo` and maybe `Foo` can be incomplete and allow for variables.

```kdl
// Template doesn't compile on its own and allows for vars being used, but other devices can copy
template Foo {
  register Blah {
    config size_bits <blah_bits>
  }
}
// This copies Foo and replaces the text `<blah_bits>` with `8`
device Foo2 ref=Foo {
  vars blah_bits=8
}
// This copies Foo, but wholly changes it
device Foo3 ref=Foo {
  remove Blah
  add {
    command Woof 5
  }
}
```

This idea is rough, but I can see something like this working. It's still a question how this should be exposed in the generated code...