diondokter/device-driver #59: esp32 xtensa target doesn't work

DivineGod, 2025-02-01:
I was trying to port the examples for the touchpad device driver I implemented in my [blog post](https://blog.mjolner.tech/device-driver-rust/) to other targets besides the rp2040 and I found that on esp32s3 (xtensa target) the device-driver crate fails to compile due to relying on `bitvec`, which in-turn relies on `radium` to supply atomics.

I found some related issues:
https://github.com/ferrilab/ferrilab/pull/2
https://github.com/ferrilab/radium/issues/11#issuecomment-1826866243
https://github.com/ferrilab/radium/pull/16

bitvec might need a bit of an update to incorporate the changes from radium 0.7 to 1.0

diondokter, 2025-02-01:
Hey, thanks for posting!
I didn't know about this...

There seems to be a PR, but it looks a bit abandoned: https://github.com/ferrilab/bitvec/pull/220
Not sure how much I can do except for forking it and using that from device-driver...

DivineGod, 2025-02-01:
I'll try that branch out and see how far I get

DivineGod, 2025-02-01:
Yes, that PR branch does indeed seem to work

diondokter, 2025-02-01:
Ok, so for now you could maybe use a `[patch.crates-io]` override for it to work. Not sure how I can fix it...
Unless you know a similar crate to bitvec that I can build device-driver on...

diondokter, 2025-02-01:
Or maybe I should implement it myself. Probably not that difficult...

closed 2025-02-08 by diondokter

diondokter, 2025-02-08:
Alright, released this as 1.0.3!
It should work fine for Xtensa now.

Took a lot of work, like 40 hours in total, but the end result is good.
Smaller binaries and fewer dependencies.

diondokter, 2025-02-08:
> Or maybe I should implement it myself. Probably not that difficult...

Lol, not true haha

PR #61 (2025-02-08): Bye bitvec
Fixes #59 

Replaced bitvec with custom logic.
Gives some nice size savings (~1.5kB or 7% for my S2-LP driver examples) and it's only a little bit slower.

Other major thing that changed is that the new custom logic operates on bytes in any byte order. Bitvec requires the buffers to all be little endian. So there's no more reversing of bytes in the generated code, so any lost performance is likely made up by that.