diondokter/device-driver #64: Feature request: Make indexing into blocks possible with enum rather than usize

rockboynton, 2025-03-20:
We typically want to select a block index by an enum rather than a usize, so current usage requires a cast to usize. 

This would allow more type safety, analogous with the rest of the crate

diondokter, 2025-03-20:
Hi, interesting idea!

Are you proposing the toolkit generates the enum?

The method itself could very easily support a `T: Into<usize>` so you can make your own.

diondokter, 2025-03-20:
But also, what would be the name of the enum and its variants?

rockboynton, 2025-03-20:
No, user defined enum, just like how we can define field type as enum

diondokter, 2025-03-20:
Maybe this can be expanded to a feature like 'named repeats'?

Now you specify the repeat count, but maybe an additional way of defining repeats could be made where you give a list of names. Then an enum with those names is generated and that can then be used to index into the block or register.

Would that work for you?

rockboynton, 2025-03-20:
I feel like just defining the enum should be enough, right? The macro knows how many variants there are

diondokter, 2025-03-21:
Ah ok cool!
Yeah I can see this work. Let's see if I can do anything for that.