diondokter/device-driver #66: Panic if output file is in a directory that doesn't exist

asasine, 2025-03-21:
If the CLI is called with `--output foo/bar.rs`, and the directory `foo/` doesn't exist, the program panics

```
device-driver-cli-1.0.4\src\main.rs:60:9:
Could not write the output to: foo/bar
stack backtrace:
   0:     0x7ff715610f01 - <unknown>
   1:     0x7ff715626ada - <unknown>
   2:     0x7ff71560e857 - <unknown>
   3:     0x7ff715610d45 - <unknown>
   4:     0x7ff71561241f - <unknown>
   5:     0x7ff715612182 - <unknown>
   6:     0x7ff715612b6f - <unknown>
   7:     0x7ff7156129f9 - <unknown>
   8:     0x7ff71561163f - <unknown>
   9:     0x7ff7156125fe - <unknown>
  10:     0x7ff7156383d1 - <unknown>
  11:     0x7ff715227b0c - <unknown>
  12:     0x7ff71522a806 - <unknown>
  13:     0x7ff71522a6e0 - <unknown>
  14:     0x7ff715608edc - <unknown>
  15:     0x7ff71522a6ac - <unknown>
  16:     0x7ff71562d8e0 - <unknown>
  17:     0x7fffbbbae8d7 - BaseThreadInitThunk
  18:     0x7fffbd83bf6c - RtlUserThreadStart
```

Version: 1.0.4

diondokter, 2025-03-21:
Thanks for the report! This indeed isn't checked

asasine, 2025-03-21:
Would it make sense to automatically create parent directories?

diondokter, 2025-03-21:
I've checked with a colleague and we found tools generally don't do this.
So a better error message makes sense, but creating the directory probably doesn't.

asasine, 2025-03-21:
Sounds good. I improved the error message to hint the user that this is a failure case in #68

PR #68 (2025-03-27): Added CLI support to write to stdout
If the `--output <FILE>` option is not specified, the CLI will write the output to stdout. This closes #65.

I also improved the error message when creating the output file when `--output <FILE>` is specified to hint the user towards checking if the containing directory exists, which resolves #66.