diondokter/device-driver #71: Make data-type in *Interface traits generic

Psy-Kai, 2025-04-17:
Hello,

I think it would be nice to have the possibility to set a custom data type when implementing the Interface traits. The fixed `&[u8]` makes it hard to use e.g. a 16-Bit bus efficiently. If you are on a little endian system you cannot transfer data in big-endian 16-Bit without buffering in the Interface implementation.  
I tried using the `core::slice::alinge_to` method to cast the `&[u8]` to `&[u16]`. Since this method only reinterprets the buffer, on little endian systems I get
```rust
let buf_u8: &[u8] = &0x0102.to_be_bytes();

let (_, buf_u16,_) = buf_u8.align_to::<u16>();

assert_eq!(buf_u16[0], 0x0201); // but I wanted 0x0102 o.o
```

It would be nice if I just could get the right data-type as input.

Psy-Kai, 2025-04-17:
Haha, no sorry, I am just stupid and forgot how SPI worked 🙈

diondokter, 2025-04-17:
Ha yeah, I was wondering which u16 bus you were using. Good to know the current design works for you in the end.

There's also a small list of drivers that use this toolkit: https://diondokter.github.io/device-driver/#known-drivers-using-the-toolkit
Could be good to look at for inspiration maybe.

Psy-Kai, 2025-04-17:
I confused myself because I used `embedded_hal_async::spi::SpiBus<u16>` and just didnt thought about just replacing the `u16` with an `u8`...

PS: thanks for the fast reply and this nice crate :)

Psy-Kai, 2025-04-25:
I think I have got a use-case for defining a custom `data` type in the *Interface trait impls:  
the STM32U5 SPI RDY pin management depends on the SPI data frame size (the `DSIZE` value in the `SPI_CFG1` register). In my case the data frame size has to be 16-Bit. This leads to the problem described in my initial message.

FYI: I am using embassy which sets the `DSIZE` value automatically. Embassy also configures the DMA channel which means I cannot twist the bytes via DMA.

diondokter, 2025-04-25:
Can you not just transmute the u8 slice to a u16 slice?

It's gonna be quite hard to make the data type generic.

diondokter, 2025-04-25:
Also, do you have a device where this actually matters?

Psy-Kai, 2025-04-25:
> Can you not just transmute the u8 slice to a u16 slice?

The transmute would interpret the buffer `[0x01, 0x02]` as `[0x0201]` instead of `[0x0102]`.

> Also, do you have a device where this actually matters?

I am currently developing without hardware and can only refer to datasheets. But the programming documentation for my target chip, the [IT8951](https://www.loboris.eu/ESP32/IT8951_SPI_Programming_Guide.pdf), requests to check the RDY signal after every transferred frame.

I could try to work around that by setting the byte-order to LE in the driver definition yaml. But I dont know if this would lead to other problems.

diondokter, 2025-04-25:
> setting the byte-order to LE in the driver definition yaml

That's what it's there for. But note that that only affects fieldsets (like registers and command in/out) and not buffers.

Worst case is that you have to convert to u16 manually and do byte swaps.
That sounds bad, but probably is not an issue. And also that's what they seem to do in their own driver: 

[image]

diondokter, 2025-04-25:
> The buffers passed to the write calls are not mut

Ah! That *is* something that can be easily change (though it would be breaking).
Let me know if `LE` does the trick. If not we can make the change.

Psy-Kai, 2025-04-28:
`LE` does work.

But it feels a little bit hacky because:
1. you have to use unsafe functions to be able to transfer a > 8Bit SPI data frame
2. the driver will not work on big-endian systems (even if most ARM MCUs will be little-endian)

diondokter, 2025-05-01:
> `LE` does work.

Great!

> 1. you have to use unsafe functions to be able to transfer a > 8Bit SPI data frame

Not amazing I guess, but not the worst...

> 2. the driver will not work on big-endian systems (even if most ARM MCUs will be little-endian)

I'll write down that the next breaking change should present the buffer as mut. That will fix that.

In the mean time, I'm open to more support for this. So feel free to hack on it.

diondokter, 2025-05-16:
Alright, closing this in favor of #78 as I think that's the better solution

closed 2025-05-16 by diondokter