diondokter/device-driver #74: CLI parsing manifest leads to panic

elpiel, 2025-04-21:
Due to lengthier error message during parsing, there's an addition of a new line `\n` and main.rs panics trying to remove the prefix for the error - `strip_prefix("::core::compile_error!(\"")`

```
$ device-driver-cli -m device.yaml -o src/ll2.rs --device-name Device

thread 'main' panicked at /home/elpiel/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/device-driver-cli-1.0.4/src/main.rs:69:14:
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

With the addition of `unwrap_or` it's clearly visible why it panics: `::core::compile_error!(\n    \"` which doesn't match the strip_prefix string:

```
$ device-driver-cli -m device.yaml -o src/ll2.rs --device-name Device
Error: "::core::compile_error!(\n    \"Parsing object `DOWNLOAD_POSTMORTEM`: Parsing error for 'fields_in': Parsing field 'LENGTH': Unexpected key: 'default'\",\n)\n"
```

diondokter, 2025-04-21:
Ah! Thanks for reporting :)

diondokter, 2025-04-21:
If you see more stuff, please let me know! Can't catch everything myself sadly :)

PR #75 (2025-04-21): Error cleanup
Fixes #74
closed 2025-04-21 by diondokter