diondokter/device-driver #76: Allow for commands to carry dynamic data

diondokter, 2025-04-21:
Currently commands can only carry a fixed amount of data.
This means commands like `read_flash` and `write_flash` can't be implemented using them.
This is annoying and should be fixed.

There should be a way for a command to carry at least 1 `&[u8]`. This comes with a whole bunch of lifetime and borrowing issues that need to be resolved eventually.

diondokter, 2025-09-26:
This may need a breaking change to the command interface. So let's take a look at this before 2.0

asasine, 2026-09-04:
What about allowing the command argument to be a DST with a slice field? It feels like it would be an ergonomic fit here.

diondokter, 2026-09-06:
> What about allowing the command argument to be a DST with a slice field?

Not quite sure what you mean. Where?

asasine, 2026-09-07:
If the input arguments type was a [DST](https://doc.rust-lang.org/reference/dynamically-sized-types.html), you could allow callers to pass in a reference to one into the generated `dispatch_in*` methods. The existing command interface already accepts a byte slice `&[u8]`, which a DST with a slice tail can be projected into, so that wouldn't need to be changed.

Conceptually, an unsized input argument might look like this:

```text
[fixed header][dynamic payload...]
```

or this in DDSL:

```ddsl
fields-in: fieldset WriteFlashIn {
    field header 31:0 -> uint,
    field data dynamic,
}
```

The generated type may look like this:

```rust
#[repr(C)]
pub struct WriteFlashIn<T: ?Sized = [u8]> {
    header: LEU32,
    data: T,
}
```

and can be constructed from any contiguous type that coerces to a slice (any [unsized coercion](https://doc.rust-lang.org/reference/type-coercions.html#unsized-coercions)):

```rust
let input = WriteFlashIn {
    header: U32Le::new(0x1234_5678),
    data: [0xAA, 0xBB, 0xCC],
};
```

the bytes are as expected in this instance:

```text
78 56 34 12 AA BB CC
[ header  ] [ data ]
```

The plumbing might look like this:

```rust
pub unsafe trait WireFieldset {
    const METADATA: FieldsetMetadata;
    fn as_bytes(&self) -> &[u8];
}

unsafe impl WireFieldset for WriteFlashIn<[u8]> {
    const METADATA: FieldsetMetadata =
        FieldsetMetadata::new().with_byte_order(ByteOrder::LE);

    fn as_bytes(&self) -> &[u8] {
        // SAFETY:
        // - LEU32 and [u8] both have alignment 1.
        // - repr(C) means no inter-field padding.
        // - LEU32 consists entirely of initialized bytes.
        // - the trailing [u8] consists entirely of initialized bytes.
        // - size_of_val(self) covers the four-byte prefix and trailing slice.
        unsafe {
            core::slice::from_raw_parts(
                core::ptr::from_ref(self).cast::<u8>(),
                core::mem::size_of_val(self),
            )
        }
    }
}
```

Going from the generated type to the command interface:

```rust
impl<'a, B> WriteFlashOperation<'a, B>
where
    B: Block,
    B::Interface: CommandInterface,
{
    pub fn dispatch_in(
        self,
        input: &WriteFlashIn,
    ) -> Result<(), <B::Interface as CommandInterfaceBase>::Error> {
        self.block.interface().dispatch_command(
            self.address,
            input.as_bytes(),
            &WriteFlashIn::METADATA,
            &mut [],
            &FieldsetMetadata::DEFAULT,
        )
    }
}
```

Callers can use either static data or an array:

```rust
// array
device.write_flash().dispatch_in(&WriteFlashIn {
    header: LEU32::new(0x1000),
    data: [0xAA, 0xBB, 0xCC, 0xDD],
})?;

// or even byte-string literals
device.write_flash().dispatch_in(&WriteFlashIn {
    header: LEU32::new(0x1000),
    data: *b"flash contents",
})?;
```

There's a small limitation here, in that Rust doesn't coerce any slice into `[u8]`, so you can't do this

```rust
// fails: WriteFlashIn<&[u8]> does not coerce to WriteFlashIn<[u8]>
let payload: &[u8] = get_payload();
device.write_flash().dispatch_in(&WriteFlashIn {
    header: LEU32::new(0x1000),
    data: payload, // ❌
})?;
```

I think that's fine, since the semantic drivers that build on top of device_driver probably won't expose this DST directly. Instead, they offer a API instead that builds a DST internally:

```rust
pub struct Flash<I> {
    device: GeneratedDevice<I>,
}

impl<I> Flash<I>
where
    I: CommandInterface,
{
    pub fn write<const N: usize>(
        &mut self,
        address: u32,
        data: [u8; N],
    ) -> Result<(), I::Error> {
        // ✔️: coerces to WriteFlashIn<[u8]> but has a sized argument to callers
        let command = WriteFlashIn {
            header: LEU32::new(address),
            data,
        };

        self.device.write_flash().dispatch_in(&command)
    }

    pub fn write_page(
        &mut self,
        page: u16,
        data: [u8; PAGE_SIZE],
    ) -> Result<(), I::Error> {
        // also coerces
        let command = WriteFlashIn {
            address: U32Le::new(u32::from(page) * PAGE_SIZE as u32),
            data,
        };

        self.device.write_flash().dispatch_in(&command)
    }
}
```

diondokter, 2026-09-07:
Ah, interesting! I don't often think of unsized types...

I think this is clever. But there's some points where I don't know if it's gonna work:
1. We're putting a lot of data through stack-allocated arrays that can be moved a lot accidentally. Is LLVM going to optimize that all out or is this going to generate a lot of memcpy's?
2. I still have the ambition to support other languages like C and C++. This same trick would work in C using `Flexible Array Members`, but C++ doesn't support that. So that's annoying since C++ would need to be solved differently.
3. What if an interface impl needs to be able to make a distinction between the fixed and dynamic parts? I guess that data could be added to the fieldset metadata?
4. I'm not sure if common traits like the embedded-storage::NorFlash can be implemented on top of this. Those traits use an `&mut [u8]` arg which can't be passed to the unsized command without stack allocating an array that's hopefully large enough and copying things over

For that last one, maybe this model is just one of two ways to represent dynamic data? We'd still need another one where a byte slice can be passed.

asasine, 2026-09-08:
1. You're not limited to stack arrays; any unsized coercion is valid, including `Box<WriteFlashIn<[u8; N]>` -> `&WriteFlashIn<[u8]>` or `&'static WriteFlashIn<[u8; _]>` -> `&WriteFlashIn<[u8]>`. The former is going to be niche for embedded systems, but combines well with [`heapless::pool::boxed`](https://docs.rs/heapless/latest/heapless/pool/boxed/). The latter is probably more common and works with `StaticCell`/`ConstStaticCell`
    ```rust
    static IN: StaticCell<WriteFlashIn<[u8; 3]>> = StaticCell::new();
    let in: &'static mut WriteFlashIn<[u8; 3]> = IN.init(WriteFlashIn {
        header: some_header(),
        data: some_fixed_sized_data(),
    });

    // coerces to WriteFlashIn<[u8]>
    driver.write_flash().dispatch_in(in);

    // can also initialiaze with const
    // all zeros should be placed in .bss and take zero stack or flash
    static IN: ConstStaticCell<WriteFlashIn<[u8; 3]>> = ConstStaticCell::new(WriteFlashIn {
        header: LEU32::ZERO,
        data: [0x00, 0x00, 0x00],
    });
    
    let in: &'static mut WriteFlashIn<[u8; 3]> = IN.take();
    fill_with_data(in);
    driver.write_flash().dispatch_in(in); // still coerces
    ```
2. If my read on [the reference](https://doc.rust-lang.org/reference/dynamically-sized-types.html#r-dynamic-sized.pointer-types) is correct, a pointer type to a struct with an unsized tail (including a reference `&WriteFlashIn<[u8]>`) is a wide pointer with metadata equal to the metadata of the struct's unsized tail. This unsized tail is a slice, so the metadata is the number of elements. Passing to C or C++ (or any/every other language) is just passing a wide pointer. I believe the metadata refers to the size of the slice, not the size of the entire struct, so exactly how to handle that wide pointer in various languages may vary, but I don't see any fundamental limitations with passing references/pointers to DSTs to C/C++/other.
3. I'd probably solve that with a conversion to a type that has the shape you're after:
    ```rust
    struct FixedAndDynamic<'a> {
        fixed: &'a [u8],
        dynamic: &'a [u8],
    }

    impl<'a> From<&'a WriteFlashIn<[u8]>> for FixedAndDynamic<'a> {
        fn from(input: &'a WriteFlashIn<[u8]>) -> Self {
            let base = core::ptr::from_ref(input).cast::<u8>();
            let dynamic = &input.data;
            let dynamic_start = dynamic.as_ptr();

            // SAFETY:
            // - `input` is a single allocation/object containing the fixed prefix
            //   followed by the trailing `[u8]`.
            // - `dynamic_start` points inside the same object, at the start of the
            //   unsized tail field.
            // - Therefore the distance between `base` and `dynamic_start` is the
            //   fixed-prefix byte length.
            let fixed_len = unsafe { dynamic_start.offset_from(base) as usize };

            // SAFETY:
            // - `base` points to the start of `input`.
            // - `fixed_len` covers only the initialized fixed prefix before `data`.
            // - The returned slice is tied to `input`'s lifetime.
            let fixed = unsafe { core::slice::from_raw_parts(base, fixed_len) };

            Self {
                fixed,
                dynamic,
            }
        }
    }
    ```
4. I think you're right in that an API taking `&mut [u8]` requires there to be an existing allocation that's sufficiently sized. IMO that's an unfortunate API but a reasonable one for embedded; the API can't handle a copy-on-write or extend mechanism like `alloc::vec::Vec` could, or a visitor pattern if the data could be handled in a streaming fashion with a limited window. Mind you, that doesn't require it to be a stack allocation (see `StaticCell` or `heapless::pool::boxed` above), and if you know the size, and your type meets the [`bytemuck::NoUninit`](https://docs.rs/bytemuck/latest/bytemuck/trait.NoUninit.html) safety requirements, you can create an initialized `&mut WriteFlashIn<[u8; N]>` using either `StaticCell` or a stack variable, cast it to a byte slice (manually via `unsafe` or via `bytemuck::bytes_of_mut(_)`), pass that `&mut [u8]` to the `ReadNorFlash::read(offset, &mut bytes)`, and your `&mut WriteFlashIn<[u8; N]>` will be filled and further coercible to `&WriteFlashIn<[u8]>` for `dispatch_in(_)`
    - Note: `bytemuck::NoUninit` requires `Sized`, but that's a design restriction in bytemuck, not a fundamental limitation of `repr(C)` as far as I can tell. `WriteFlashIn` does meet the requirements otherwise, and so I believe `bytemuck::bytes_of_mut(_)` could support DSTs and remain sound.

--- 

> For that last one, maybe this model is just one of two ways to represent dynamic data? We'd still need another one where a byte slice can be passed.

I suppose the point I'm trying to convey is that, for dynamic data, a plain byte slice and and a DST with an unsized slice tail (a "slice DST" maybe?) are fundamentally equivalent for device_driver fieldsets. The assumption here is that fieldsets are all already `NoUninit + AnyBitPattern` (technically `Pod`) because they're all based on `[u8; _]` arrays, and that composes even when nesting these types, so treating instances as byte slices as slice DSTs all boil down to the same thing. Slice DSTs afford unique ergonomics since they give you named fields that map into named offsets in that slice.

diondokter, 2026-09-08:
1. It's weird though to statically allocate the whole fieldset instead of just the data buffer. But that can be ok. The only other thing is that `dispatch_in(in)` doesn't work. It'd be `dispatch_in(|val| *val = *in)`, so that'd be a copy. An extra function could be added that's not closure-based, but that'd be a bit annoying.
2. My concern is not passing the data from Rust to C++, but generating C++ code instead of Rust code. So the ddc would produce C++ code and then it's not possible to create this in a non-hacky way (and without allocator)
3. I mean, you might need to know the boundary in the command interface impl. There the fieldset types don't exist anymore, it's just a byte slice. But I can image an interface that needs to have a difference in behavior between the fixed and dynamic part. For example, an I2C device might take the command, then need a repeated start before accepting the data. So the interface needs to be able to know about that. The metadata could have that info, but not for bulk combined fieldsets. But I think that's solvabele.
4. Yeah it's very common in embedded Rust. So common that I am hesitant to create something that doesn't work well with it.

For reference, since I didn't spell it out in my issue opener, this is sort of the way I originally though how this feature should work roughly:
```ddsl
command Foo {
    address: 0,
    dynamic-data: allow, // Opt into having dynamic data
    fields-in: ... // Still have fields
}
```
```rust
device.foo().dispatch_in(|fields| /*...*/, &mut my_buffer)?;
```
```rust
trait DynamicCommandInterface {
    fn dispatch_dyn_command(
        &mut self,
        address: Self::AddressType,
        input: &mut [u8],
        _input_metadata: &FieldsetMetadata,
        output: &mut [u8],
        _output_metadata: &FieldsetMetadata,
        dynamic_data: &mut [u8], // <- new, would just be passed the `my_buffer` from the user
    ) -> Result<(), Self::Error>;
}
```
I think this has none of the issues I'm raising. Though I do like it being part of the fieldset. That's pretty neat. It just maybe doesn't solve the problem I originally spotted which prompted me to open this issue.

Is there a chip where your design would excel over my design?