diondokter/device-driver #89: cfg on Bitfield leads to compiler error on generated file

janikwitzig, 2025-05-25:
I am writing a driver for the nRF24L01x family of devices. There is a field that used to exist for the nRF24L01 but does not in the nRF24L01+, so I wanted to lock it behind a feature.

The relevant section from the `json` is:
```json
"Rf_setup": {
        "type": "register",
        "description": "RF Setup Register",
        "access": "RW",
        "address": 6,
        "size_bits": 8,
        "reset_value": 7,
        "fields": {
             {...},
            "lna_hcurr": {
                "cfg": "not(feature = \"nrf24l01p\")",
                "description": "Setup the LNA Gain. 1 -> High Gain, 0-> Low Gain",
                "base": "bool",
                "start": 0,
                "end": 1
            }
        }
    },
```

which removes the `lna_hcurr` field when the `nrf24l01p` is selected.

This has the following effects in the generated file:
```rust
///Read the `lna_hcurr` field of the register.
///
///Setup the LNA Gain. 1 -> High Gain, 0-> Low Gain
#[cfg(not(feature = "nrf24l01p"))]
pub fn lna_hcurr(&self) -> bool {
    let raw = unsafe {
        ::device_driver::ops::load_lsb0::<
            u8,
            ::device_driver::ops::BE,
        >(&self.bits, 0, 1)
    };
    raw > 0
}

///Write the `lna_hcurr` field of the register.
///
///Setup the LNA Gain. 1 -> High Gain, 0-> Low Gain
#[cfg(not(feature = "nrf24l01p"))]
pub fn set_lna_hcurr(&mut self, value: bool) {
    let raw = value as _;
    unsafe {
        ::device_driver::ops::store_lsb0::<
            u8,
            ::device_driver::ops::BE,
        >(raw, 0, 1, &mut self.bits)
    };
}
```

but unfortunately  it does nothing on the `impl::fmt::Debug`:

```rust
impl core::fmt::Debug for RfSetup {
    fn fmt(&self, f: &mut core::fmt::Formatter<'_>) -> Result<(), core::fmt::Error> {
        f.debug_struct("RfSetup")
            .field("cont_wave", &self.cont_wave())
            .field("rf_dr_low", &self.rf_dr_low())
            .field("pll_lock", &self.pll_lock())
            .field("rf_dr_high", &self.rf_dr_high())
            .field("rf_pwr", &self.rf_pwr())
            .field("lna_hcurr", &self.lna_hcurr())
            .finish()
    }
}
```
resulting in this error:

```zsh
➜  embedded-nrf24l01x git:(update-device-driver) ✗ cargo build --features nrf24l01p                                                                 

error[E0599]: no method named `lna_hcurr` found for reference `&RfSetup` in the current scope
    --> src/nrf24l01p.rs:2067:43
     |
2067 |                 .field("lna_hcurr", &self.lna_hcurr())
     |                                           ^^^^^^^^^ method not found in `&RfSetup`
```

diondokter, 2025-05-25:
Ah, actually, looking at it the problem is still there. Easy fix though. Will push it soon!

diondokter, 2025-05-25:
Btw, if you have ideas for or thoughts on how chip families should be supported (like you're doing now), I'd love your feedback here: https://github.com/diondokter/device-driver/issues/58